XMegaForArduino/arduino #38: some arduino libraries cause errors

zdjurisic, 2017-04-03:
I'm just guessing that it is the cores issue... So, with that said:

Including certain libraries causes compiling errors. For example,

`#include <XWire.h>`
`#include <XRTClib.h>`
`RTC_DS1307 RTC;`
`void setup() {}`
`void loop() {}`

throws:

`test_TWI_DS1307:3: error: expected ')' before '*' token`

the XRTClib is a copy of the RTClib, in which ony the reference to Wire was changed to XWire. Same thing with a few other libraries.

zdjurisic, 2017-04-04:
I should probably mention that the same code, just <Wire> and <RTClib>, compiles fine for, say, Mega2560, or Nano or ProMini on 328 chip

bombasticbob, 2017-07-30:
added https://github.com/XMegaForArduino/libraries/issues/6

(this issue left for reference; please post additional comments over in 'libraries')