XMegaForArduino/arduino #39: Clock set feature to add

zdjurisic, 2017-04-21:
XMega is the first AVR that can set the clock at runtime, rather than burning a fuse. There should be a code (maybe in a form of a library?) that can set the source and the frequency of the clock. Let's discuss a proper way to do it within the scope of XMegaForArduino.

bombasticbob, 2017-07-30:
I've considered clock speed reduction code for power savings.  problem is that it messes up a lot of things when you do that, from baud rates to PWM to the 'millis()' timer.  Good idea, but I've pretty much decided that it's too much for a microcontroller.  However, no problem doing an alternate 'core' that supports it, maybe.

basically anything that uses 'F_CPU' to determine the CPU frequency will be affected.  And changing it means you have to go back and re-initialize things like serial baud rates while comms are still happening.  So yeah, it's a complex thing.

bombasticbob, 2017-07-30:
one more thing, it's a good idea, so please don't take it wrong if it really can't be done (easily, without wasting a lot of code space for everything else, etc.).