XMegaForArduino/arduino #43: program  chip with Arduino IDE

mkeyno, 2018-07-04:
Hi
I own the ATXMEGA128A3U Dev. board that I used Atmel Flip to program it , but I don't know how to program it with Arduino IDE, any help is really appreciated 
BR

bombasticbob, 2018-09-08:
@mkeyno 

depending on your bootloader, you should be able to use the 'wiring' protocol like an mega2560 uses [or the bootloaders I've done for 128A1[U] and 128A4U.  However, if you don't have a compatible bootloader, then you'll probably need to use a PDI-capable programmer that's also supported by Arduino's IDE.  Sorry if that's bad news.  I use AVR ISP Mk II but that's no longer made by ATMel as far as I can tell.  They have another (more expensive) programmer, though, so hopefully that one works with avrdude [I have not yet tried it].

bombasticbob, 2018-11-05:
added some additional support for 128A4U, similar to the 128A1U for the bootloader.

(bootloader still needs USB version)